Rocket-Show/rocketshow #10: Add default composition with image with instructions on how to proceed

moritzvieli, 2020-09-20:
For people setting it up with a monitor. The black screen after startup is a little confusing.

DavidOpgh, 2023-02-06:
Yes, this was a bit confusing at first. Maybe a default composition with the install would helpful for beginners. But I found the default composition setting and I inserted my own.

DavidOpgh, 2023-02-22:
@moritzvieli now that I'm playing with Designer it would be helpful to have a very basic default Designer Project linked to the default composition.

DavidOpgh, 2023-04-24:
Instead of the black screen after startup how about a simple image that says "Rocket Show"?
I would think that would be pretty simple to implement. 

Maybe offer the user the option to add their own splash screen image?

During development it would be helpful to me to have a simple image that tells me Rocket Show has started.

DavidOpgh, 2023-05-06:
@moritzvieli 

It would be helpful if there was an option to toggle the "black screen after startup" with the default being set to off. 
Having the monitor show status information is very helpful in the early development stage. 
When the show is completed is when "black screen after startup" is useful.