Rocket-Show/rocketshow #16: Change volume of the whole composition or each separate audio/video files

moritzvieli, 2020-09-20:


DavidOpgh, 2023-02-06:
This would be a nice feature to have. The ability to normalize the audio levels of each composition. Where you could change the levels of each audio file in a composition and then the overall level of the composition.

rennist, 2023-07-06:
I second this feature as well.

DavidOpgh, 2024-10-08:
@moritzvieli 
Any thoughts about including this feature in a future release?

rennist, 2024-10-08:
I'm still very interested.... just a little horizontal volume slider per file or even a  +/- db field to set would be good.   Currently when I find I have recorded a channel/track to hot for the mix then I have to go back to Cubase and adjust it then re-export the file and import again into RS.   I am only outputting 4 channels  (Click, MONO and Stereo Backing Tracks for the mains) so I can't adjust anything specific in the Stereo tracks with the mixer.  Having RS control over the audio files' levels would be ideal.

moritzvieli, 2024-10-29:
@ideadapt please check https://github.com/moritzvieli/rocketshow/commit/d1938518874a4f8ac8e965875d98f3152c30e2f6

I added the ability to adjust the volume of single audio files. It's released to the test channel. Volume of the composition cannot yet be adjusted (i think, we just need to multiply it with the volume of a single file).

moritzvieli, 2024-10-29:
Upon 2nd thought, I would not allow to change the volume on the whole composition. It might be confusing, because a composition includes audio/video/midi files and the volume is only applied to audio files. If this requirement comes up in the future, we would need to discuss it.

Maybe @DavidOpgh or @rennist can quickly check the new feature on the test branch (available on the update page in newer Rocket Show versions). If everything is fine, I'll release this feature officially.

@ideadapt fyi

DavidOpgh, 2024-10-29:
I hope you reconsider adding the function to allow the overall volume to be changed on the whole composition in addition to the stems. Because both are needed to solve two different issues.

The ability to change the volume of the stems allows you to get the correct mix for a song.

The ability to change the volume of the composition allows you to get the correct level for all the songs in the show.

I don't think it's confusing. 
Any video in a composition shouldn't be an issue because any audio imbedded in the video file isn't played.
I would think it would be a help if someone has midi files in addition to audio files in the composition.

moritzvieli, 2024-10-29:
@DavidOpgh ok, pushed to the test branch. maybe you could take a look at it. 👍

DavidOpgh, 2024-10-29:
OK thanks. It might take me a day or two but I'll take a look at it.

rennist, 2024-10-30:
@moritzvieli   I just did the update to the test 2.4.2 and it looks like it's working well.    tested a few different audio files and the Master volume for the composition and it all works great.   I checked other basic functionality to make sure nothing was broken as far as I can tell and it was all good.
Thank you for this functionality, it will be very helpful... now we just need a <PLAY> button from the Composition screen for testing.  Cheers !

DavidOpgh, 2024-11-11:
I did a simple test of the beta with the default media and it appears to work well. 

I would agree with @rennist it would be helpful if there were transport buttons available on the Composition page where you could test out the volume changes instead of switching back and forth from the Composition and Start pages.

DavidOpgh, 2024-11-12:
@moritzvieli - I'm probably being picky but I'm wondering whether a logarithmic db scale would be better than a linear % scale. It seems there isn't a significant change in volume until the % gets pretty low. @rennist any thoughts about this or is % scale good enough?

moritzvieli, 2024-11-18:
@DavidOpgh a logarithmic scale seems to make sense indeed. found this article and implemented it accordingly with x^4:
https://www.dr-lex.be/info-stuff/volumecontrols.html

moritzvieli, 2024-11-18:
@rennist @DavidOpgh I added this issue regarding playing a composition directly in the editor: https://github.com/moritzvieli/rocketshow/issues/126

With this issue, we will now only check the volume controls and close, if it works fine. I'll let you know, as soon as the change with the logarithmic scale is ready on the test channel. (y)

DavidOpgh, 2024-11-18:
@moritzvieli - thanks for the update.

moritzvieli, 2024-11-24:
@DavidOpgh @rennist  logarithmic volume change is now live on the test channel. let me know, if it's still not satisfactory.

DavidOpgh, 2024-11-24:
@moritzvieli - I'll check it out!

moritzvieli, 2024-12-02:
@DavidOpgh Had to read the article a couple more times but I think, yes, it needs to be dB now! will be fixed with the next release! :)

closed 2024-12-02 by moritzvieli