Rocket-Show/rocketshow #19: On the Raspberry Pi 4 there is a bcm headphone audio device, but it does not work

moritzvieli, 2021-02-05:
Gstreamer error:
`2021-01-21 07:44:53.998  INFO 2432 --- [io-8080-exec-10] c.a.r.c.CompositionPlayer                : Playing composition 'Audio'...
0:00:07.095915533  2432 0x90ac3cc0 WARN                    alsa pcm_hw.c:1711:snd_pcm_hw_open: alsalib error: open '/dev/snd/pcmC0D0p' failed (-22): Invalid argument
0:00:07.096355607  2432 0x90ac3cc0 WARN                    alsa pcm_dmix.c:1165:snd_pcm_dmix_open: alsalib error: unable to open slave
0:00:07.096881144  2432 0x90ac3cc0 WARN                    alsa gstalsasink.c:848:gst_alsasink_open:<audiosink> Error -22 (Invalid argument) calling snd_pcm_open (&alsa->handle, alsa->device, SND_PCM_STREAM_PLAYBACK, SND_PCM_NONBLOCK)
0:00:07.097383070  2432 0x90ac3cc0 WARN                    alsa gstalsasink.c:865:gst_alsasink_open:<audiosink> error: Could not open audio device for playback.
0:00:07.097704496  2432 0x90ac3cc0 WARN                    alsa gstalsasink.c:865:gst_alsasink_open:<audiosink> error: Playback open error on device 'rocketshow': Invalid argument
2021-01-21 07:44:54.049 ERROR 2432 --- [ervice thread 1] c.a.r.c.CompositionPlayer                : GST error: Could not open audio device for playback.`

Playing it with `aplay -D rocketshow xy.zwav` works fine. Seems to be a Gstreamer issue. Maybe update Gstreamer and try again?

moritzvieli, 2021-02-05:
If no solution: Deactivate the headphone audio device.

moritzvieli, 2021-02-21:
Temporarily deactivated

DavidOpgh, 2023-02-05:
So will there ever be support to output audio from the headphone jack on the Pi 4 ?

moritzvieli, 2023-02-05:
@DavidOpgh I need to check again with the new version. However, we had complaints about the quality in the past anyway. Seems like the headphone jack is not well suited for public audiences. I would therefore anyway recommend a separate audio interface for serious applications.

DavidOpgh, 2023-02-05:
Yes, I understand the audio from the headphone jack would be inferior to a
separate audio interface but while developing in the prototype phase it
would be helpful to have the option.

moritzvieli, 2023-02-05:
Yes, that‘s true, that could be useful. I‘ll check it and let you know. 👍