Rocket-Show/rocketshow #53: Not working with Raspi 4b 8gb

Inkell67, 2022-11-14:
It says something like start4.elf is out of date

moritzvieli, 2022-11-15:
@Inkell67 Thanks a lot for this issue. Can you add some more details? Where does it say this exactly (how can we reproduce it)? What is the error exactly? Could you maybe add a screenshot?

Inkell67, 2022-11-18:
Hi Moritz, that's what I see booting

moritzvieli, 2022-11-20:
@Inkell67 can you please provide the detailed error/warning/information or a screenshot/photo? thanks a lot!

Inkell67, 2022-11-20:
[image]
[image]

Inkell67, 2022-11-20:
[image]

Inkell67, 2022-11-20:
[image]

moritzvieli, 2022-11-20:
@Inkell67 This looks like the image it's not properly flashed. Which tool did you use to flash it?

Inkell67, 2022-11-20:
[image]

moritzvieli, 2022-11-20:
@Inkell67 Ah, I see. Please don't create separated answers next time. We'll look into it. Thanks a lot.

Inkell67, 2022-11-26:
Any suggest to make it work? Maybe copying start4.elf and other from a fresh DietPi distro?

hacker-hippo, 2023-01-31:
I am also experiencing this issue!

moritzvieli, 2023-01-31:
@Inkell67 @hacker-hippo I need to setup a new environment to build a new image. I am trying to do it asap and let you know.  (y)

hacker-hippo, 2023-01-31:
Boots on Raspberry Pi 3, fails on Raspberry Pi 4.

moritzvieli, 2023-02-04:
@Inkell67 @hacker-hippo There is a new image ready to test here: https://rocketshow.net/install/images/2023-02-04-RocketShow-2.2.5.zip

Could you try it out and let me know, whether it works with the new devices? Unfortunately, I don't have one here to test this issue. Thanks a lot for your feedback!

DavidOpgh, 2023-02-04:
2.2.5 is not running on my 4GB Pi 4 but 2.2.3 does

moritzvieli, 2023-02-04:
@DavidOpgh If the display goes dark, it means Rocket Show has loaded (in case you connected a projector or something, black is the idle screen if no video is played). It should go dark in the 2.2.3 version as well. Do you see the wifi "Rocket Show" published? Can you connect to it and open "http://rocketshow.local"?

DavidOpgh, 2023-02-04:
I'm familiar with 2.2.3 when it is loaded and i initially  thought 2.2.5
rocket show was loaded but that wasn't the case. My first clue was no
rocket show wifi network to connect to. The pi 4 screen goes dark and no
other activity. The included screenshot happens very quickly, probably less
than 10 seconds from initial boot. I have the boot sequence on video which
I can post if needed.

moritzvieli, 2023-02-05:
@DavidOpgh Ah I see, yes, no signal is not good for sure. Strange. On the initial boot the SD card gets extended and an automatic reboot follows. I guess, the device is not awaking afterwards again, correct? Do you have a possibility to connect it to a network with an ethernet cable and SSH into it?

DavidOpgh, 2023-02-05:
You are correct. After the reboot the device does not wake again.

I do have the ability to connect it to a network with an ethernet cable but
I'm unfamiliar on how to use SSH.
Is this a good guide?
https://www.makeuseof.com/how-to-ssh-into-raspberry-pi-remote/

For the command line
ssh ***@***.***.local
What would be the [username]?

I'm assuming [hostname].local would be rocketshow.local or would it be
better to locate the ip address?

If I'm successful to SHH into the Pi 4 is there something I should do?

David

DavidOpgh, 2023-02-05:
The entire boot sequence of 2.2.5 on my 4gb Pi 4
https://drive.google.com/file/d/14ISvzKlWcrjOaDI_3rEnv52_1mBh2UE2/view?usp=share_link

moritzvieli, 2023-02-05:
@DavidOpgh Thanks a lot for your tests! I will download the image and test again on my side. Maybe the upload failed or there's something else. Yes, you need to insert the SD card again in your computer after flashing it and create a file called "ssh" in order to open the SSH connection. But if you don't even get an IP with the new image, we don't need to try SSH. Yes, if Bonjour service works (usually on all Apple devices, on Windows not always), rocketshow.local  instead of the IP should also work. But let me compare first, how it should look like.

moritzvieli, 2023-02-12:
@DavidOpgh I just downloaded the provided version and burned it with the Raspberry Pi Imager and everything seems fine. Can you maybe try it with this tool instead of Win32diskimager? This is the new recommended imager for Raspberry Pi and maybe something was added to the image builder (pi-gen) which requires this tool.

Thanks a lot for your feedback!

DavidOpgh, 2023-02-12:
> @DavidOpgh I just downloaded the provided version and burned it with the Raspberry Pi Imager and everything seems fine. Can you maybe try it with this tool instead of Win32diskimager? This is the [...]

@moritzvieli Yes, using Raspberry Pi Imager to burn the image worked correctly. 

You're welcome. I'm excited to see everything RS can do.

moritzvieli, 2023-02-13:
@DavidOpgh Thanks a lot for the confirmation. I just updated the website and changed the recommended tool from Win32diskimager to the Raspberry Pi Imager. Also, I released the new image. Therefore, I consider this issue closed.

closed 2023-02-13 by moritzvieli

Inkell67, 2023-02-14:
Thank you Moritz!