Rocket-Show/rocketshow #62: What DMX interfaces are supported?

DavidOpgh, 2023-02-06:
Is there a list of supported DMX interfaces?

I have a DMXIS usb interface and it's recognized by OLA but I'm having issues with it communicating with my Chauvet Gigbar2 in the 23 channel mode. The Gigbar2 responds correctly in the 23 channel mode when I connect it to my standalone DMX controller. I'm trying to isolate where the problem might be. 

One big problem is there doesn't appear to be a good fixture configuration for the Gigbar 2. The 3 ch mode seems to configured correctly but not the 11ch or 23ch. And running the OLA DMX console isn't working correctly. 

I also have a DMXKing eDMX 1 Pro but I haven't been able to figure out how to configure it with Rocketshow yet, or if it's even possible. OLA doesn't seem to be recognizing it. I'm guessing I need to give it a static IP address.

moritzvieli, 2023-02-06:
@DavidOpgh You can find the list of supported devices here: https://www.openlighting.org/ola/

Unforutnately, I don't see DMXKing eDMX 1 Pro listed here, so seems like it's not supported. There is a link in the Rocket Show settings under lighting, which opens the OLA admin page including a manual controller, where you can fiddle around with the channels. However, if you don't see a universe automatically created by Rocket Show, it does not seem to work.

closed 2023-02-06 by moritzvieli

DavidOpgh, 2023-02-06:
Thanks for the info.

DavidOpgh, 2023-02-08:
I was able to get the DMXIS interface working correctly when I connected it through a powered USB hub.

DavidOpgh, 2023-02-10:
This $25 interface works if you enable the FTDI DMX USB plugin in OLA
https://smile.amazon.com/Interface-Converter-Computer-Lighting-Controller/dp/B07D6LNXF9/ref=cm_cr_arp_d_product_top?ie=UTF8&th=1

DavidOpgh, 2024-08-27:
Just for reference these are the DMX interfaces I've tested with Rocket Show on PI 5

Cheap $25 interface from Amazon (works if you enable the FTDI DMX USB plugin in OLA):
https://smile.amazon.com/Interface-Converter-Computer-Lighting-Controller/dp/B07D6LNXF9/ref=cm_cr_arp_d_product_top?ie=UTF8&th=1

Enttec DMXIS (OLA recognizes this as "Enttec Usb Pro Device"
Enttec DMX USB Pro 
Enttec Open DMX USB (works if you enable the FTDI DMX USB plugin in OLA)

DMXking ultraDMX Max (works if you  add this line to ola-usbserial.conf: "device_prefix = ttyACM"