Rocket-Show/rocketshow #65: OLA plugin config issue - how to disable plugin

DavidOpgh, 2023-02-08:
I'm having trouble getting an Enttec Open DMX USB device to operate in RS.

It appears OLA is incorrectly identifying the Enttec Open DMX USB device as a StageProfi device since that is the plugin it's trying to use. 

OLA reports the Enttec Open DMX plugin is loaded and active.
How can I  disable the StageProfi plugin as a test to see if there is a conflict?

DavidOpgh, 2023-02-10:
I figured out how to disable plugins, although I couldn't figure out how to list all of the plugins with their ID#. I had to look in the log file to figure out which id went with which plugin.

My test didn't turn out like I expected.
When I disabled the StageProfi plugin the Enttec Open DMX interface wouldn't connect at all. 
The Enttec Open DMX plugin didn't recognize the interface.

As another test I bought a cheap $25 USB to DMX interface off of Amazon
https://smile.amazon.com/Interface-Converter-Computer-Lighting-Controller/dp/B07D6LNXF9/ref=cm_cr_arp_d_product_top?ie=UTF8&th=1

I had to disable the Enttec Open DMX plugin and enable the FTDI USB DMX plugin.
It connected successfully and it appears to be working properly in OLA.

Here is the command

NAME
ola_plugin_state - Get and set the state of the plugins loaded by olad.
SYNOPSIS
ola_plugin_state --plugin-id <plugin-id> [--state <enable|disable>]
DESCRIPTION
ola_plugin_state is used to get and set the enabled/disabled state for a plugin and the list of plugins this plugin will conflict with.
OPTIONS
-h, --help
Display the help message
-p, --plugin-id <plugin_id>
Id of the plugin to fetch the state of.
-s, --state <enable|disable>
State to set a plugin to.

moritzvieli, 2024-07-15:
@DavidOpgh I found out, you can enable/disable plugins in the new OLA interface, which is still in beta unfortunately. Can you try the following steps?

1. Move to Rocket Show Settings
2. Lighting tab
3. Open Lighting Console
4. On the very bottom, click "New UI (Beta)"
5. On the top, click "Plugins"

Would this solve your issue without having to ssh into the raspberry pi?

Thanks a lot for your reply!

DavidOpgh, 2024-07-15:
I'll try out the beta UI and report back.

I've moved away from using ssh to configure the OLA plugins and now recommend using an attached USB keyboard to type the commands.

DavidOpgh, 2024-07-15:
> you can enable/disable plugins in the new OLA interface, which is still in beta

@moritzvieli 
The OLA beta UI only works on my PI5, not on my PI4.
On the PI5 I can confirm you can enable or disable plugins through the new beta UI (although you still have to know which plugins to enable/disable because some interfere with others)

There seems to be something missing on the PI4 which affects how the hyperlinks on the beta UI page are being handled. 

For example the home page link is 
http://rocketshow.local:9090/new/#/
but on the PI4 the browser is returning 
http://rocketshow.local:9090/new/#!/#%2F
Although the browser shows the new home page it appears anything after the first # in the address is not being processed correctly. 

For example the plugin page link is
http://rocketshow.local:9090/new/#/plugins
but browser is returning
http://rocketshow.local:9090/new/#!/#%2Fplugins
and is only showing the home page, so there is no way to access the plugins on a PI4

[image]

moritzvieli, 2024-07-22:
@DavidOpgh Thanks a lot for your test. As the new beta is still in beta, such issues could occur. On the Pi 5, compared to the 4, newer dependencies (such as OLA, see https://github.com/OpenLightingProject/ola) are included as well. This is the reason, why it works on the 5 but not on the 4.

Except for critical updates however, I am not building new images for older Raspberry Pis, because it takes quite some effort (and dependencies might be incompatible). The process is documented here, if you want to do it by yourself: https://github.com/moritzvieli/rocketshow?tab=readme-ov-file#raspberry-pi-image-building

But because this issue is fixed for the Pi 5, I would close it, if you agree. Maybe you could even add a section in your startup guide?

DavidOpgh, 2024-07-22:
Thanks for the clarification. 
So although Rocket Show should run on a Pi 4 it would be difficult to recommend using it now because these newer dependencies might never be corrected. I'll note this in the start up guide and add a section about the beta OLA plugin interface.

DavidOpgh, 2024-07-22:
I agree #65 can be closed

closed 2024-07-22 by DavidOpgh