Rocket-Show/rocketshow #67: Leadsheet - documentation or example?

DavidOpgh, 2023-02-12:
I would like to know more about this feature of Rocketshow. 

Can the leadsheet be a pdf or must it be a jpg?

moritzvieli, 2023-02-16:
@DavidOpgh Also here, we need to create some documentation, you're right.

No, currently only images (jpg, png, etc.) are supported. But this feature has been requested already in the past and we'll look into it.

DavidOpgh, 2024-11-12:
I played around with the leadsheet function. 
@moritzvieli - if you use this function can you explain what it's trying to accomplish?

what I observed
I added a jpg to a composition and it was pretty straightforward. It showed a document icon for the composition after refreshing the browser.

You need to click on the icon for the photo to be seen. A Close button is shown to close the photo.

The photo appeared to be autosized to fill the width of the screen and is not resizable  , so the screen ratio is important if you want the entire photo to be seen. 

I only tried it on my laptop running RS in the browser. 

Where I could see this being useful. 
If a 2nd external monitor was connected to the laptop it would be nice if this leadsheet would automatically be shown on the external monitor when the composition was started.