Rocket-Show/rocketshow #68: Backup / Restore function?

DavidOpgh, 2023-02-12:
I would like the ability to backup and restore my compositions / media.

Any plans to add that functionality?

Any work around available to do this manually?

moritzvieli, 2023-02-14:
@DavidOpgh Currently, there's no such feature implemented. It has been requested however already a couple of times. I'll keep this issue therefore open.

As a workaround, users usually just clone the SD card and keep the image to flash it back in case something goes wrong.

rennist, 2023-07-06:
I would like to see some sort of a backup/restore as well.... I am using a 128 Gb card and even though I currently only have less than 10 Gb files on it, it take almost 45 minutes to back it up to an image file.    It would be great to have the ability to send the image directly from Rocket Show or even better to be able to Export complete Compositions and setlists like you can Projects from Designer.   Each time I change or add anything in a Composition then I need to do another backup image.
Thanks

DavidOpgh, 2023-07-11:
@rennist 
What software are you using to create your image files?
Thanks!

rennist, 2023-07-11:
@DavidOpgh    On my Windows PC I use Win32DiskImager .... on my Macbook I just use the Disk Utility and create an image on the Hard Drive from the entire SD Card (not just the Boot).   It works but it takes a while because it images the entire 128 Gb, bit by bit.

DavidOpgh, 2023-07-11:
@rennist 
Thanks for the reply. 
I'm at the point where I need to start backing up my SD card because I've put in a lot of time into making presets / scenes in Designer.

rennist, 2023-07-11:
@DavidOpgh   I learned the hard way... I tried running the update on RocketShow (I think to version 2.3.1) and then I was unable to get RocketShow to start after that with some exceptions in the log.... I didn't have time to troubIeshoot that since I had rehearsal that evening so I had to scramble and restore my latest backup image, but that image was missing some of the latest compositions and it still took almost an hour to restore.  It would have been great if I had the missing Compositions saved as well to restore them after the image restore.

DavidOpgh, 2023-07-11:
@moritzvieli 
Thanks for the info. I'll keep that in mind. That should work at the point I'm at now.
But I'll still want a complete backup SD card on hand when I create the entire show.

moritzvieli, 2023-07-11:
@DavidOpgh Sure! I agree. Also to have a backup at hand to have the whole setup redundant in case something breaks during the show. Or also for @rennist 's case, of which I'm sorry to hear!

DavidOpgh, 2023-07-11:
@moritzvieli 
Now that I'm working in Designer I'm finding it would be helpful if it was possible 

to copy scenes and presets (ex. create a new preset that only changes the color of an existing one).

and be able to use those presets / scene for other Projects files

rennist, 2023-07-19:
@DavidOpgh ... I have only used that program for an initial Flash but I will try it for Cloning the SD Card next time.  Thanks !

moritzvieli, 2023-08-02:
Chunking needs to be implemented first in order to allow the upload of large files (> 2 GB) for the restore function:
https://stackoverflow.com/questions/49769853/dropzone-js-chunking

DavidOpgh, 2023-08-02:
@moritzvieli 
Thanks for the update. I wonder if anyone using media files that large with RS? 
I know mine are much smaller than 2GB.
If it could be implemented sooner I could live with a Backup/Restore function with that limitation.

moritzvieli, 2023-08-02:
@DavidOpgh The backup file will contain all media files combined. Our project already has much more than 10 GB with all videos, etc. I think, chuking really is a requirement.

DavidOpgh, 2023-08-02:
@moritzvieli 
Thanks for the clarification. 
What are you using to backup your project?
The cloning function on the Balena Etcher program using fast memory cards is working well for me.

moritzvieli, 2023-08-02:
@DavidOpgh I am using the on-board tools on the Macbook: https://blog.jaimyn.dev/the-fastest-way-to-clone-sd-card-macos/

DavidOpgh, 2023-08-02:
@moritzvieli 
Thanks for the info.

moritzvieli, 2024-07-13:
@DavidOpgh @rennist I just pushed a new version to update existing rocket show devices with a new feature to create and restore backups. Please let me know, if it does not work as expected. Thanks a lot!

closed 2024-07-13 by moritzvieli

DavidOpgh, 2024-07-13:
@moritzvieli Can I use this to migrate from my show on the PI4 to my new PI5.

moritzvieli, 2024-07-14:
@DavidOpgh yes, that should work.