Rocket-Show/rocketshow #74: Designer - explanation of Scene and Preset parameters

DavidOpgh, 2023-02-21:
Can you give a quick explanation of these parameters.

Scene parameters:
Fade In
Before
Fade Out
After

Preset parameters:
Start
End
Fade In
Before
Fade Out
After

moritzvieli, 2023-03-28:
@DavidOpgh The fade in/out parameters define, how long a a scene or preset "smoothly merges" into the scene or preset before or after. It's a linear factor that defines, how "strong" an effect applies (from 0 to 100%).

if you check "after", the fade out will start fading out right after the position (same with before).

As mentioned by email, there was an issue when working with real hardware. I just pushed an update (version 2.2.5). Please try again with this vrsion.

closed 2023-03-28 by moritzvieli

DavidOpgh, 2023-03-28:
So version 2.2.5 dated 2023-02-04 now contains an update instead of
issuing a new version?

moritzvieli, 2023-03-28:
@DavidOpgh  Yes, I have not built a new image yet. It's just an online update currently for all existing installations.

DavidOpgh, 2023-03-28:
Thanks for the clarification.

DavidOpgh, 2023-04-01:
I sent you email. The update to fix fade in/out is still not working for me.
I downloaded the 2.2.5 image from the website and installed it.
Everything else seems to be working fine.
I created both a scene and a preset with a fade in of 6000ms and neither worked.
The virtual fixture shows the fade in but the actual fixture on comes on at full brightness after 6000ms.
I've included my Designer project
[DTestDMX.zip](https://github.com/moritzvieli/rocketshow/files/11130053/DTestDMX.zip)

moritzvieli, 2023-04-02:
@DavidOpgh There is no new image. Did you download the the jar-file directly and installed it on the device? The auto-update feature should detect the new version and install it, when connected to the internet. But I will build a new image soon anyway.

DavidOpgh, 2023-04-02:
Moritz
Thanks for the clarification.

I wasn't aware there was an auto update feature.
Even though the device is connected to the internet I saw no indication it
detected a new version or tried to install one.

When you stated "It's just an online update currently for all existing
installations" I tried using the manual "update" button but it didn't
detect a new version. I still have the memory card with that version. I'll
boot the image and check again.

Since I didn't see any indication the device detected an update, that's
when I tried downloading the latest version, mistakenly thinking the
changes were incorporated in the image, even though you explicitly stated
it wasn't. This new build also didn't seem to detect an update.

I'm not very knowledgeable on Raspberry Pi software so I wasn't aware
Rocket Show is using Java to even look for a jar file.
If you walk me through the steps where the updated jar file is located and
where it needs to be installed I'll try that.

I'm looking forward to a new image. Let me know if I can assist. I think
I've documented my requests for future versions.

David

DavidOpgh, 2023-04-05:
@moritzvieli 
Can I get a brief explanation about these Preset parameters?
Start
End

DavidOpgh, 2023-04-08:
Moritz
So is there any update on why the auto update didn't work?
Is there anything else I can do except wait for a new image release?
thanks
David

DavidOpgh, 2023-04-16:
Moritz
Can I get a quick reply as to what's happening with the auto update or when
we can see a new image?
thanks!
David

moritzvieli, 2023-04-16:
@DavidOpgh Not sure why it does not work. Maybe I messed up the versioning. Could you please try again whether you see the update now? Best,
Moritz

DavidOpgh, 2023-04-16:
Moritz
What should I see if the update works correctly?
David

moritzvieli, 2023-04-16:
@DavidOpgh Go to the settings:
[image]

Click search for updates and you should be able to update to the latest version:
[image]

DavidOpgh, 2023-04-16:
Mortiz
It must have been an issue with the versioning.
When you changed it to 2.2.6 the auto update now works.
I'll test out the changes this week and let you know what I find.
Thanks!
David

DavidOpgh, 2023-04-17:
And to really close out this issue can I get a brief explanation what these preset parameters do and how they are used.?
Start
End

Thanks!

moritzvieli, 2023-04-17:
Thanks a lot for the log finding. I just fixed it (and set the correct level) for the next release.

And yes, regarding the two parameters start/end: They define, where a preset starts or ends within the scenes. If you want to place e.g. two presets one after another into the same scene, you can adjust these two parameters accordingly.

Please let me know, if it's unclear. Thanks a lot!

DavidOpgh, 2023-04-17:
OK - thanks for the explanation. I believe I understand it. 
You would only use either the Start or the End parameter in a Preset for a Scene, never both. 
(if you tried using both would it/should it generate an error?)
I'll test it out and report back if I have any issues. 

The only thing I'm unclear on concerning this issue is how do to a partial fade. 
i.e. 
a linear fade from 100% to 50% in 5 seconds. 
I'm guessing this would have to be done as part of a Scene?

moritzvieli, 2023-04-22:
@DavidOpgh No, both parameters can be used together. They act like a "blocker" inside the scene. A preset can start somewhen e.g. 2'000 ms after the scene starts and stop 1'000 ms before the scene ends.

If I wanted to fade from 100 to 50 % in 5 seconds inside a scene (and the scene lasts e.g. 15 seconds), I would add two presets: One which "keeps" the base light on a level of 50 % all the time, and one preset which starts with 100 % and goes down to 0 % with a fade out of 5 seconds and an end position of 5'000 ms as well. I just tested it and in this case, it fades down from 100 % to 50 % within 5 seconds inside the scene.

DavidOpgh, 2023-04-22:
@moritzvieli/rocketshow
***@***.***>

OK - Thanks for the clarification about Start / End

Thanks for the explanation about the partial fade:
I will have to test this. Maybe it will make sense when I try it.
I'm still having a hard time wrapping my head around the parameters.
In my head a preset doing a linear fade of 100% to 0% in 5 seconds would be
at the 50% level in 2.5 seconds, not 5 seconds.
And the 50% "keeps" preset would keep the light from going any lower than
50%

David