Rocket-Show/rocketshow #76: Designer - Fixture Pool does not include a new profile uploaded locally as part of a Project

DavidOpgh, 2023-02-21:
The Fixture pool in Designer is strictly a copy of the profiles available from the Open Fixture Library? 
OFL didn't have the profile for my fixture so I created my own using the OFL Fixture Editor but I didn't submit it, I only downloaded locally. The new profile is in my Project but is not available in the Fixture Pool.

Is there any way I can add my own fixtures to the Fixture Pool?

moritzvieli, 2023-02-22:
@DavidOpgh Yes, Rocket Show relies on the OFL. And yes, there is a possibility to upload your own fixtures created with the OFL editor after downloading the OFL JSON. But I just noticed, there is a bug preventing this function currently (the fixture is just not shown, as you mentioned). This will be fixed soon. 

Anyway: We strongly recommend submitting new fixtures to the OFL in order to support this great project. But it might take some time until we update all fixtures again from the OFL to make them available for Rocket Show by default.

DavidOpgh, 2023-02-22:
Yes, I thought about submitting it but it wasn't a very good implementation, just something to test the fixture, so I decided not to submit it at this time.

moritzvieli, 2023-03-28:
Just checked this issue again. I tried with the wrong JSON. One needs to download the real OFL JSON, not the draft "work-in-progress" JSON. If you use the OFL JSON, it works as designed. @DavidOpgh with which JSON did you try?

DavidOpgh, 2023-03-28:
From your Youtube video I believe you recommended downloading the JSON
before submitting to OFL because of the delay before it becomes available
there. If that is the work-in-progress JSON then that's the process I used..

Thanks for verifying using the real OFL JSON works as designed.

moritzvieli, 2023-03-28:
@DavidOpgh Correct. You need to click "create fixture" -> "preview fixture" and then there's a download button for all formats:

[image]

This step is before submitting it to the OFL.

Did you try with this JSON?

DavidOpgh, 2023-04-19:
@moritzvieli 
I retried your procedure 
Clicked the "create fixture"  then "preview fixture" and then downloaded the json. (I did submit the fixture this time)
My fixture is still not part of the fixture pool.
It only exists in the project where I used the "Create from profile file"
It doesn't show up in the fixture pool when I create a new project.

I have included the json and my project file

[ofl_export_hsl_40w-beam-spot-light-rgbw_ofl.zip](https://github.com/moritzvieli/rocketshow/files/11278309/ofl_export_hsl_40w-beam-spot-light-rgbw_ofl.zip)
[Test1.zip](https://github.com/moritzvieli/rocketshow/files/11278312/Test1.zip)

moritzvieli, 2024-07-06:
@DavidOpgh sorry for the late reply! That's the designed behaviour. Each project contains a "copy" of the included fixtures from the pool (this also protects the project from broken shows, if a fixture gets updated for example). If you include an own fixture, it's also only valid for the added project (it's just added to the project's copy of fixtures). If you want the fixture to be added to Rocket Show globally, you need to submit it officially to the OFL. Otherwise, you need to import it for each project separately.

closed 2024-07-06 by moritzvieli

DavidOpgh, 2024-07-06:
Thanks for the explanation. It makes sense although when was the last time Rocket Show globally updated the fixture pool from the OFL? I've submitted my fixtures to OFL and have never seen Rocket Show include it or any other recent approved fixture submissions to OFL.

moritzvieli, 2024-07-06:
@DavidOpgh good point. I have not updated it for some time now already, because only few people use the designer currently. Also, the display of the lightrays broke some time ago with a Chromium update. I fixed it and a update including the new fixtures will be available soon.

DavidOpgh, 2024-07-06:
Thanks for the confirmation. An update and some designer fixes would be welcome. Using the current designer has been a challenge. I'm working on a new version of the Start Up guide and documenting the designer has been interesting.