Rocket-Show/rocketshow #77: My Rocket Show Startup Guide - comments welcome

DavidOpgh, 2023-03-07:
Rocket Show Startup Guide by DavidOpgh
Version 1.0.20230307

Installing Rocket Show on a Pi4 isn't difficult but there are some important steps you need to know.
I wrote this guide so you could benefit from my efforts and to help you get up and running quickly.
Note: This is very long and detailed. It might look difficult but it's not, it's just very comprehensive.

Installing Rocket Show on Pi4 (mine is the 4GB version)

Plug all your devices into the Pi4 (see Appendix A for a list of my devices)

Download latest version of Rocket Show and unzip the image to your PC 
https://rocketshow.net/install/images/latest.php

Download and Install Raspberry Pi Imager (currently Version 1.74) on your PC
https://www.raspberrypi.com/software/

Insert microUSB card into your PC
I like these microUSB cards, they have a Speed Class of C10, V30, and U3(4). Anything lower than this class are noticeably slower.
https://www.amazon.com/SanDisk-Extreme-microSDHC-UHS-3-SDSQXAF-032G-GN6MA/dp/B06XWMQ81P

Run Raspberry Pi Imager
Click "Choose OS" button
Scroll to the bottom and select the "Use custom" option
Navigate to the location on your PC where you unzipped the Rocket Show image file and select it.

Click "Choose Storage" button.
Select your microUSB card

/* this next step is not absolutely necessary if you're planning to have 
a usb keyboard connected to the Pi 4 to directly access the command prompt.
However I highly recommend enabling SSH to allow remote access from the laptop.
Because as you will soon see you will most likely need to change some settings. 
See Appendix B about SSH access */
Click gear options button on the Raspberry Pi Imager
Click checkbox "Enable SSH"
Set username as rocketshow
Set password as thisrocks
Leave all other settings as the defaults
Click "Save" button

Click the "Write" button
Click "Yes" button to continue. 
When completed Windows will pop up a windows saying you need to format the USB memory card before using it. Click "Cancel"
Raspberry Pi Imager will pop up window saying you can remove the SD card. Click "Continue"
Exit Raspberry Pi Imager

Remove the USB card from the PC and insert it into Pi4
Boot Pi4
Console terminal will pop up a message saying Generating SSH Keys
Console terminal will pop up a message saying rebooting in 5 seconds
Pi4 automatically reboots
Console terminal will show some text spelling out the words Rocket Show and the version number.
Console terminal screen goes blank. This means Rocket Show has now started.
(you can wake up the Console terminal screen with any key press on the connected USB keyboard)

SSH login 
/* If you haven't read Appendix B about SSH login you should do it now. */
/* If you haven't read Appendix B.1 about enable/disable OLA plugins you should do it now. */

/* If you have a USB keyboard connected to your Pi4 you can enable/disable OLA plugins
through the USB keyboard after logging in locally to Rocket Show on the Pi4. See Appendix B.1 */

At this point you should have enabled / disabled the correct OLA plugins.
If not go to Appendix B.1

Now that you have the correct plugins enabled, reboot the Pi4 for the changes to take effect.

Log in remotely to the Pi4 from your computer using the address rocketshow.local on your browser.

[... 237 lines cut ...]

Plugin StageProfi --- Id=8
Plugin Enttec Open DMX --- Id=6 (Yes, I know it's surprising that this plugin doesn't work with the ENTTEC OPEN DMX USB interface)
Plugin Serial USB --- Id=5

To disable these plugins type in these commands
ola_plugin_state -p8 -sdisable
ola_plugin_state -p6 -sdisable
ola_plugin_state -p5 -sdisable

Here is the 1 plugin that needs enabled
Plugin  FTDI USB DMX --- Id=13

To enable this plugins type this command
ola_plugin_state -p13 -senable

Note: If you need to find the id of a plugin
There might be an easier way but this is how I did it.
You can do this after Rocket Show is up and running.
Go to the OLA Admin browser tab page in your browser
rocketshow.local:9090/ola.html 
Click on plugin name located in the column on the left side 
Click the "View Log" link located in upper right corner of the page
Look for a recent entry in log for the command
Sending request [GET json/plugin_info?id=XX where XX will be the Id of the plugin you clicked on.

moritzvieli, 2023-03-08:
@DavidOpgh Wow! Highly appreciated, well done! I am thinking on where such guides could be hosted. I think it will be something like markdown files hosted here on GitHub and linked from the website. I'll come up with something and let you know. 👍

DavidOpgh, 2023-03-08:
@moritzvieli 
You're welcome. Glad to contribute. Looking forward to exploring more features of Rocket Show.

moritzvieli, 2023-03-30:
@DavidOpgh I prepared the folder docs (currently only on develop-branch) with a getting-started.md markup file. If you want, you could add your guide there and make your first "official" repository contribution. Otherwise I'll transform your input here into markdown. Let me know, what you prefer. (y) :)

DavidOpgh, 2023-03-30:
I would like the guide to have the ability to be updated in the future so
which method would be better?
For example the original guide was formatted for Github and currently
doesn't contain any photos which I think would be helpful to include.

My original plan was to update this google doc
https://docs.google.com/document/d/1qc-VDSoUHncOlFBArfpUA7hCuyIxQjfIOHbg5ceMGdw/edit?usp=sharing
But I'm open to implementing whatever process you think would work best.
If there is a markdown editor that I could use to make the process easier
I'm open to that.

moritzvieli, 2024-07-16:
Closed in favour of https://github.com/moritzvieli/rocketshow/issues/111

closed 2024-07-16 by moritzvieli