Rocket-Show/rocketshow #80: Designer live preview setting - under what conditions would you not want to enable it?

DavidOpgh, 2023-03-10:
The title says it all. 
The System.Lighting page contains the Designer live preview option.
Why would you ever want to not enable it?

moritzvieli, 2023-03-23:
@DavidOpgh This is one scenario I could imagine: Your Rocket Show device is hooked up to a lighting rig at a location where already some people are present. You want to make some quick changes but you don't want to push them for everyone to see. Albeit this is most probably a rare case.

Also, the live preview could clash with running compositions (e.g. a different user starts a composition while you're still editing a scene, or you have a 2nd browser tab open and don't close the designer).

I would suggest doing two things to reduce confusion here:
- Enable the live preview by default (already done, not yet released)
- Automatically disable live preview while playing a composition (to do)

DavidOpgh, 2023-03-24:
@moritzvieli 
Thanks for the explanation.

moritzvieli, 2024-09-08:
@DavidOpgh Indeed, thanks for the link. In this case I'll close this issue.

moritzvieli, 2024-09-08:
Sorry, wrong issue