Rocket-Show/rocketshow #81: Midi Out - Controlling Devices

rennist, 2023-03-11:
Good day.... I am very impressed with Rocket Show so far but I cannot figure out how to Send Midi messages out to a device (change patches, effects, banks, etc. on my Line6 effects board).  Do I need to add the Line6 board as a Fixture in Designer and if so how? Then I assume I would have to create a different Preset/Scene for each specific Control Change of Preset Change?

Any instruction would be appreciated.

Thanks,

Rick

DavidOpgh, 2023-03-14:
I haven't tried using midi with Rocket Show yet. 
But I believe you would create a midi file and add it to the Composition that would "play" the messages you want sent to your Line6.

rennist, 2023-03-14:
Hey, thanks David, I didn't think of that but it makes sense.   I'll create a Midi File in my DAW and see if that works.

moritzvieli, 2023-03-23:
@rennist @DavidOpgh Exactly, this is how it should work. I'll close this issue. Please let us know, if there are still issues with the MIDI.

closed 2023-03-23 by moritzvieli