Rocket-Show/rocketshow #82: Audio Error when including video in Composition

rennist, 2023-03-17:
I am unable to include any video in a composition and play it, receiving this error every time.   I can play compositions with just multiple audio tracks but as soon as I add an mp4 to the composition then it fails (including the example composition.
[image]

Any assistance is appreciated.

Rick

rennist, 2023-03-18:
apparently never mind... it is working now.  Not sure why the error came up before every time but it's good now.

Thanks.

DavidOpgh, 2023-03-21:
I'm not experiencing the problem you're having. I'm able to save 1080 video with my compositions.  The only issue I was having Initially was glitchy playback but I corrected it when I changed the encoding to CBR at 4Mbps

moritzvieli, 2023-03-23:
@rennist Could you provide a faulty MP4 file? Seems like Gstreamer has some issues decoding the file. Maybe I can find some more information when debugging it.

rennist, 2023-03-23:
Hi Moritz... I have attached a smaller file that also does not work.  Same error. Thanks for looking into it. 

https://user-images.githubusercontent.com/104380717/227362676-12ee17be-5194-4b11-9e84-49ba1f8a1bf0.mp4

DavidOpgh, 2023-04-19:
I'm have the same issue. Doesn't matter if it's 2.2.5 or 2.2.6
Here is a link to a sample file. 
https://drive.google.com/file/d/15gDqwd52J0b6VBD9hxNrF1sWpPEmNT4k/view?usp=share_link

moritzvieli, 2023-05-05:
@DavidOpgh @rennist A new update is available (connect your device to the internet and search for updates). This will solve this issue: There will still be a timeout, but with the new Gstreamer version it seems, with certain file types, there just is a timeout although the data is discovered correctly.

A new image will be available soon.

DavidOpgh, 2023-05-05:
Wrong photo in previous post[image]

DavidOpgh, 2023-05-05:
I tried updating to 2.3.1 from a working version of 2.2.6 and Rocketshow threw the same exception and hung.

moritzvieli, 2023-06-21:
@DavidOpgh If you try with the new image, does it work then? From what I tested, it should be fine now.

closed 2023-06-22 by moritzvieli