Rocket-Show/rocketshow #85: gstreamer error with default composition using 2.2.6

DavidOpgh, 2023-04-19:
Looks like whatever changes you made to gstreamer for 2.2.6 has somehow affected the playing of the default composition. In 2.2.6 the video file plays correctly as a stand alone composition. But when selected as the default composition it throws an error which also causes RS to hangs and can't be recovered.

I tested on three different systems

I tested it with 2.2.5 and the same video file has no issues as the default composition. 

Here is the error message on the console.

[image]

DavidOpgh, 2023-04-19:
[image]

moritzvieli, 2023-04-22:
@DavidOpgh Thanks a lot for the report! There was indeed an update of the Gstreamer base libraries with the new image. I need to take a detailed look into it.

As a workaround yes, you can remove the default composition in the command line: Go to the folder `/opt/rocketshow` and open the settings file using `nano settings.xml`. Now you need to remove the line with `<defaultComposition>xy...` and save the file. If you restart Rocket Show, it should be gone.

DavidOpgh, 2023-04-22:
@moritzvieli 

Thanks the info. 
Is the Gstreamer update also related to issue #82 ?

moritzvieli, 2023-05-05:
@DavidOpgh Yes, this issue seems to be resolved as well with #82.

closed 2023-05-05 by moritzvieli