Rocket-Show/rocketshow #86: can VNC server be used with Rocketshow?

DavidOpgh, 2023-04-22:
To make it easier to access the Pi4 I wanted to remote in from my laptop using a graphical interface. 
I enabled the VNC Server on the Pi4 using sudo raspbi-config and installed VNC Viewer on the laptop.
But when I connect using VNC Viewer gives me a message saying it can't display the desktop.

It possible to use the GUI with Rocketshow?

DavidOpgh, 2023-04-22:
I went in another direction and installed WinSCP on my laptop that gives me the functionality I was looking for to access the files on the Pi 4

moritzvieli, 2023-06-22:
In general, Rocket Show is also compatible with the desktop version of the pi OS, but there is no dedicated image provided. You would need to install the dependencies and the app manually.

Glad you found a suitable way for you.

closed 2023-06-22 by moritzvieli