Rocket-Show/rocketshow #93: Bug - Rocket Show's Fixture Pool JSON interpreter doesn't appear to support matrices

DavidOpgh, 2023-07-07:
Having a major issue trying to get the Chauvet DJ - Gigbar 2 fixture from the fixture pool to work in Designer. 
Tried using the Fixture Pool entry and downloading the JSON directly from the Open Fixture Library. 
Neither worked correctly

The fixture has 3 modes: 23-channel, 11-channel, 3-channel.
Only the 3-channel mode had the proper configuration for all 3-channels. 
The 11-channel didn't show any channels.
The 23-channel only showed a couple of channels.

Looking at the JSON in Notepad it appears the issue is Rocket Show interpreter doesn't support matrices. 
The 3-channel mode doesn't use any matrices to configure the channels
the 11-channel mode uses all matrices
the 23-channel mode uses mostly matrices.

@moritzvieli can you verify this behavior?
Can this behavior be fixed or should I just make a custom Gigbar 2 fixture JSON by removing the matrices?

Description
https://github.com/OpenLightingProject/open-fixture-library/blob/master/docs/fixture-format.md#matrices

moritzvieli, 2024-08-29:
@DavidOpgh I just published an updated version supporting matrix fixtures. This is quite a large update for the designer and could introduce some side-effects. Also, there is a migration being done between old designer projects and new ones (done automatically, when you open an old project with the new version and save it). I specifically tested with the Gigbar 2, but I don't own the fixture physically. Can you check, whether it works as expected?

Thanks a lot for your feedback!

DavidOpgh, 2024-08-29:
Yes, I will test it with my Gigbar 2. Can you tell me a little more about the auto migration? What does it try to do?

moritzvieli, 2024-08-29:
Just some changes in the internal data structure of the designer project files. Fixtures are stored a little bit different now, having fixtures with multiple beams.

DavidOpgh, 2024-08-29:
[image]

[image]

[image]

First test results
On a Pi 4 I upgraded to 2.40

The designer project gave the window saying the project was updated. I checked out some of my gigbar2 presets from my custom profile and they worked properly. I tried running the composition without saving and it threw that error. 

After saving the project and restarting the composition it worked properly with no noticeable error.

DavidOpgh, 2024-08-29:
2nd testing results

I'm not sure what the rules are for implementation but what you're using needs to be refined

Here is your implementation of the par and derby for gigbar2

[image]

[image]

And my implementation 
[image]

[image]

moritzvieli, 2024-08-30:
@DavidOpgh great! Thanks a lot for the findings!

I need to check the error from the upgrade. Seems like something has not synced properly between the frontend and the backend.

The other errors regarding unreachable devices seem unrelated (maybe slave devices shut down?).

Regarding the 2nd test with radio buttons: I noticed it during my tests with the Gigbar as well, but did not investigate further yet. This is unrelated to the matrix fixtures, but it needs to be fixed as well. From what I saw, it‘s three options and one of them has a dmx range and should hence display a slider additionally, if selected, to adjust the value. This is just a small fix and I will look into it soon as well. Thanks for confirming! 👍

DavidOpgh, 2024-08-30:
The unreachable error is unrelated and it not really an error.  I normally have a 2nd Pi running Rocket Show but it was disconnected for the test.

DavidOpgh, 2024-08-30:
> This is unrelated to the matrix fixtures, but it needs to be fixed as well. From what I saw, it‘s three options and one of them has a dmx range and should hence display a slider additionally, if [...]

This is issue #94

moritzvieli, 2024-09-08:
@DavidOpgh Indeed, thanks for the link. In this case I'll close this issue.

closed 2024-09-08 by moritzvieli