Rocket-Show/rocketshow #94: Request - Designer - Value slider .i.e more robust implementation of DMX channels with multiple capabilities

DavidOpgh, 2023-07-08:
@moritzvieli 

I've included the configuration a "Par 1 Shutter" control channel which has 4 capabilities, 2 of which (Intensity and Shutter Strobe) have a range of values the user can select from. But Designer only implements them as 4 radio selector buttons with no ability to select a particular value. I request a "Value" slider be implemented along with the radio buttons to allow the user to select a value in that capabilities' particular range.

    "Par 1 Shutter": {
      "capabilities": [
        {
          "dmxRange": [0, 127],
          "type": "Intensity",
          "brightnessStart": "off",
          "brightnessEnd": "bright"
        },
        {
          "dmxRange": [128, 239],
          "type": "ShutterStrobe",
          "shutterEffect": "Strobe",
          "speedStart": "slow",
          "speedEnd": "fast"
        },
        {
          "dmxRange": [240, 249],
          "type": "ShutterStrobe",
          "shutterEffect": "Strobe",
          "soundControlled": true
        },
        {
          "dmxRange": [250, 255],
          "type": "ShutterStrobe",
          "shutterEffect": "Open"
        }
      ]
   },

moritzvieli, 2024-09-08:
@DavidOpgh If you navigate to the rocketshow settings, after upgrading to 2.4.0, there's a new option to select the update branch:

[image]

If enabled, you'll see the latest updates, not yet officially released. In this update, I included changes, which rework the channel view. Can you check, whether it works as expected with the ranges for the GigBar 2?

Thanks a lot for your feedback!

DavidOpgh, 2024-09-09:
I have activated the test-branch and installed the latest update. I have tested the Gigbar2 profile in Designer and have confirmed the channel view works as expected with channels that have ranges for multiple capabilities.

moritzvieli, 2024-09-09:
@DavidOpgh Great! Thanks for your feedback! I released this version officially and will close this issue.

closed 2024-09-09 by moritzvieli

DavidOpgh, 2024-09-09:
Excellent! This is going to make the Designer much more useful to users. I didn't check, but has the Fixture Pool been updated as well?

moritzvieli, 2024-09-09:
Great! Indeed. I just updated the fixture pool as well. 👍

DavidOpgh, 2024-09-09:
Excellent.  When I have a moment I'll try to update the documentation to include all the latest updates.