Rocket-Show/rocketshow #95: Anyway to hack the Designer RSD file so you can copy scenes/presets from one Project to another?

DavidOpgh, 2023-07-11:
@moritzvieli 

I'm building a lot of presets / scenes in Designer that I'd like to reuse from one Project file to another since I'm using the same fixtures.
It doesn't look like Designer has a way to copy scenes/presets.
I was wondering if I could this by hacking the RSD files.

I haven't tried either method but I can think of two ways I might be able to accomplish this.

Method 1 
Build a "template" Project RSD file with all the presets/scenes for all the fixtures you're using.
Export "template" RSD File
Hack the RSD filename so RocketShow thinks it's a different Project aks "NewSong"
Import "NewSong" RSD file
Change the composition for the new song.

Method 2
Copy these sections from one RSD file to another.
 "scenes": [
  "presets": [

Any thoughts if either method is viable or which method might be a better choice?
Thanks!

DavidOpgh, 2023-09-05:
@moritzvieli 
Can I get a recommendation which method(s) might work better?

thanks!

moritzvieli, 2023-09-08:
@DavidOpgh In general, the original idea was to create only a single project and add all the compositions to it. But sure, both methods work as well. Advantage of method 1: You don't need to edit the RSD file manually. Maybe you can start with this approach but when you need to implement updates (e.g. a new fixture), you need to copy it around anyway with method 2.

closed 2023-09-08 by moritzvieli

DavidOpgh, 2023-09-08:
@moritzvieli 
Thanks for the reply. 
Maybe I don't understand how to properly use Projects.
So a Project should be considered like a "Master Fixture Pool"?
I would sync different compositions to the same Project then use "Save As" the composition name?