Rocket-Show/rocketshow #96: Function Request - Play function on Composition screen

rennist, 2023-07-12:
Add the ability to play the composition from the Compositions screen to preview the tune with the current settings.  i.e.  I mute certain variations of tracks and sometimes depending on what band members are present but I' like to preview to confirm how it sounds.   Or when I am creating or editing a composition to select the tracks to play/mute.  Currently have to go out to the START screen, locate the composition and then play.

Thanks,

Rick

rennist, 2023-07-12:
oh, and I would LOVE a volume option per track...somehow to subtract +/- db from individual outputs/tracks.  But now I'm just being greedy.

DavidOpgh, 2023-07-13:
I don't know if you've ever considered using a secondary device for the audio but there are some Ipad apps that have the functionality you're looking for
Stage Traxx 3
Multitraker 
I believe you could trigger them from RocketShow through midi.

rennist, 2023-07-13:
@DavidOpgh ...I'm not sure if that would work.   I really just need a way to control the audio output level for each output to the soundcard (there is no mixing capability on the soundcard, just USB in and 4 outputs).  If I could have individual file level controls it would be even better since I sometimes have multiple Audio files playing out the same outputs but would be nice to be able mix them as well right from RocketShow.

DavidOpgh, 2024-11-12:
I mentioned this elsewhere but I agree with @rennist it would be helpful to have ability to play the composition from the Compositions screen to preview the tune with the current settings.

moritzvieli, 2024-12-09:
discussed and implemented with https://github.com/moritzvieli/rocketshow/issues/126

closed 2024-12-09 by moritzvieli