Rocket-Show/rocketshow #100: Some items I’d like to see in RocketShow after 2.3.2

DavidOpgh, 2023-07-24:
With the release of Rocketshow 2.3.2 which has been pretty stable for me,  there are some things I'd like to see. 

Biggest hurdle for new users for plug and play usability:
- No process  to configure the OLA drivers for DMX interfaces without using ssh.

2nd biggest hurdle for plug and play usability
- Designer - fixture pool is not properly implemented for some profiles, no implementation for matrices.

Initial startup
-  In Settings.Lighting it should automatically enable “Designer live preview” (or make this the default)

- In addition to asking for the language and name, it should take you to Settings.Audio to allow you to set the default “Audio device” and then offer to restart if changed.

- It should ask to configure OLA drivers for the DMX interface then offer to restart if changed.

- I would like to see a configurable splash screen on startup instead of a blank black screen.

Menu bar
- I would like to see a “Shutdown” button added at the top instead of having to go to Setting.System page

- I would like to see a “Reboot” button added at the top instead of having to go to Setting.System page

Settings.Network
- I would like to see the IP address displayed. 

Designer
- I would like a simple sample Designer Project using the default media and the “Generic - RGB” fixture synchronized to the default Composition. The Project would include sample Presets and Scenes using the Fade in / Fade Out / Start / End parameters and Curve Effect.

- fix issue #76 - uploaded fixture only gets added to current project - not fixture pool

- Give a warning to save the Project before attempting to move to another tab when in Designer.

- Designer would display the Project name currently loaded.

- Increase zoom on the timeline when the resolution is precision is greater than 1/8

- The ability to make a copy of a preset / scene

- A global preset / scene pool where presets / scenes can be reused in other projects.

Backup / Restore function
- The ability to completely backup Rocket Show

moritzvieli, 2024-07-13:
Backup/restore has been discussed/introduced with https://github.com/moritzvieli/rocketshow/issues/68

moritzvieli, 2024-07-15:
@DavidOpgh Regarding "No process to configure the OLA drivers for DMX interfaces without using ssh.", I remember you had an issue we discussed somewhere (by email)? But I cannot find it anymore. Do you have some more details or can you maybe create an own issue? Unfortunately, I don't remember the problem anymore or what you tried to achieve exactly. Thanks a lot!

moritzvieli, 2024-07-15:
@DavidOpgh Regarding shutdown/reboot in the menu bar: I'd rather not implement this because of the following reasons:
1. I'd like to keep the menu bar as simple as possible. Every new item makes it harder for newcomers to comprehend it.
2. Those two functions could ruin a live show, when pressed accidentally. I'd therefore avoid putting them in such an exposed position.
3. Personally, I almost never used those functions (reboot only during setup phase). But in general, the raspberry is quite stable with only cutting the power. Additionally, with the new backup feature now introduced, it should not be an issue, even if the sd card got corrupted in a seldom case.

I hope, you agree. Thanks a lot!

moritzvieli, 2024-07-15:
> @DavidOpgh Regarding "No process to configure the OLA drivers for DMX interfaces without using ssh.", I remember you had an issue we discussed somewhere (by email)? But I cannot find it anymore. Do [...]

@DavidOpgh Found an issue: https://github.com/moritzvieli/rocketshow/issues/65 was it this one?

DavidOpgh, 2024-07-15:
> "No process to configure the OLA drivers for DMX interfaces without using ssh."

I initially wanted a way to easily configure the OLA drivers without having an attached USB keyboard. But it wasn't worth the effort. In the new startup guide I removed all references to ssh. I now show how to configure the OLA drivers using the USB keyboard.

Yes, issue #65 is where I documented the issue.

I'll try the beta UI and report my findings.

DavidOpgh, 2024-07-15:
> Regarding shutdown/reboot in the menu bar: I'd rather not implement this because of the following reasons:

Currently using shutdown/reboot pops up a window for confirmation,  so I don't believe an accident would be as likely as you think. But you have a lot more experience using RocketShow than I do so and if you think it's stable enough to just cut power I'll defer to you. The shutdown/reboot is still available, just not as convenient.

DavidOpgh, 2024-07-15:
> I'll try the beta UI and report my findings.

I can confirm it works on a PI5 but not a PI4. My complete findings are in. 
https://github.com/moritzvieli/rocketshow/issues/65