Rocket-Show/rocketshow #101: Remote controlled device not selecting/playing the correct composition   ( or was it always broken?)

DavidOpgh, 2023-07-26:
@moritzvieli I'm kinda losing my mind today :( 

I have my primary Rocket Show PI4 configured to play my secondary RocketShow PI4 using the Network Remote Device settings. I thought I had it working correctly. The IP addresses being used appear to be correct. I'm able to shutdown and reboot the secondary from the primary.

I thought when configured correctly, a remotely connected Rocket Show device would select/play the same composition as the primary. 

That isn't happening today. 

Although the primary does seem able to start/stop the secondary's composition, the secondary is not playing the same composition. The secondary plays whatever composition is currently selected in the Start queue. Selecting a different composition on the primary isn't being recognized by the secondary. 

This seems like a new issue but I haven't changed anything on either PI4 recently.  So is this new or has it always been broken?

Rebooting or reestablishing the remote connection didn't correct the issue. 

I'll going to build two new images from scratch and test this further.

DavidOpgh, 2023-07-28:
[image]

This is the terminal on the remote device. The last 4 RPC Call failures happened when I selected different compositions on the master device.

moritzvieli, 2023-07-29:
Hi @DavidOpgh . Thanks for the report!

I just checked the code and found a possible issue introduced long time ago. But I am not sure, if this really solved the problem. I just released a new version (not a new image) 2.3.3. Could you maybe update your remote device online and check, whether this solves the issue?

Thanks a lot for your feedback!

moritzvieli, 2023-07-29:
@DavidOpgh Additional hint: The logs are unrelated. If you select a composition on the master device, the compositions on the remote devices are just stopped and within this step, the lighting universe is reset. Seems like no lighting device has been enabled on the remote device and therefore an RPC call fails. The new composition should be selected and played on the remote device, as soon as the composition is played on the master device.

DavidOpgh, 2023-07-29:
@moritzvieli 
Thanks for the quick reply! 
Yes, running version 2.3.3 on the remote device fixed the issue. The remote device will play the composition selected on the master device.

Thanks for info about the logs. 
Since logging appears to be broken on the latest releases the screen shot was the only feedback I could provide.

moritzvieli, 2023-07-30:
@DavidOpgh Thanks for your quick test! Yes, have not forgotten about the broken logging feature. I will take a look into it. 👍

closed 2023-07-30 by moritzvieli