Rocket-Show/rocketshow #105: Any issues if I try swapping out an audio file for a revised one?

DavidOpgh, 2023-09-05:
Before I try this I'll ask here first. Maybe one of you have already done this.

The issue
I need to swap out an audio file ( the original needs some level and compression changes ). 
I'm planning on giving the revision the same name as the original. 

What I've observed
From what I can tell in the configuration file Rocketshow only uses the filename for identification so it shouldn't know the difference if a file with the same name and same length is swapped out.  

My worry
If I try replacing the original will I lose all of the lighting cues I made in Designer. 

My thoughts
Can I do this by editing the composition, deleting the original and adding the revision? Or if I try adding a file with the same name will Rocketshow overwrite the original?
 
Or do I need to do something like export the Designer project, swap the files, Import the Designer project. 

Or would it be better to use something like WinSCP to swap out the file? That way Rocketshow has not aware of the change.

moritzvieli, 2023-09-05:
@DavidOpgh As long as the length of the file did not change, you can simply upload the new file with the same name and it will replace the old one. If the length has changed, I recommend to save the composition again for Rocket Show to recognize the change. But the designer will still work and play the cues at the same location for both scenarios.

DavidOpgh, 2023-09-05:
Thanks for the quick reply.