Rocket-Show/rocketshow #108: Any thoughts about creating a Pi 5 version?

DavidOpgh, 2024-01-16:
@moritzvieli 
Just wondering if there is a Pi 5 version in the future?

moritzvieli, 2024-05-28:
@DavidOpgh A version for the Pi 5 has been built and can be downloaded here: https://rocketshow.net/install/images/2019-06-23-RocketShow-2.1.3.zip

I'll keep this issue open for a few days for people to try out. If everything looks good, it'll be released officially soon. (y)

DavidOpgh, 2024-05-28:
Nice! - I was waiting to hear back before buying one. Can't wait to try it out. Thanks!

DavidOpgh, 2024-05-28:
I am curious whether this Pi5 version will have all the latest updates since the download image seems to suggest an older 2019 version?

moritzvieli, 2024-05-28:
@DavidOpgh Sorry, linked the wrong image. I just updated the link. It's from May this year.

DavidOpgh, 2024-05-28:
OK - thanks for the clarification.

moritzvieli, 2024-06-03:
64bit Raspberry Pi 5 version is now officially available on https://rocketshow.net/get-started/ (y)

closed 2024-06-03 by moritzvieli

DavidOpgh, 2024-06-03:
The latest Pi 4 version is 2.3.5. Is there a difference between this
release and the Pi 5 release which is 2.3.2 ?

DavidOpgh, 2024-06-03:
I think you still have the issue with linking to the wrong version. The Pi 5 release should be linked to 2.3.5

moritzvieli, 2024-06-03:
> The latest Pi 4 version is 2.3.5. Is there a difference between this release and the Pi 5 release which is 2.3.2 ?
> […](#)

There are only some minor bugfixes:
- select MIDI device, if there are multiple
- disable the WLAN AP correctly, if not used
- don't initialize the OLA-universe (and throw an error), if it's not available

DavidOpgh, 2024-06-03:
The link for the Pi 4 download seems linked to 2.3.2 instead of 2.3.5

moritzvieli, 2024-06-04:
@DavidOpgh That's correct. There's no image for 2.3.5 on the Pi 4. You have to burn the 2.3.2-image and update inside the app. We usually don't maintain a image for the last version on all Pis, but only on the newest model.

DavidOpgh, 2024-06-07:
minor issue - splash screen on startup shows version 2.3.6

DavidOpgh, 2024-07-06:
Can you confirm the PI 5 version is 2.3.5 since the splash screen shows 2.3.6.
Were there any updates included in PI 5 version? I thought I saw there were some minor fixes. 
When I tried to update my PI 4 through RS it wouldn't let me update.

DavidOpgh, 2024-07-06:
Let me clarify. My PI 4 version is version has been 2.3.5 for a while and Rocket Show won't let me update to pick up these bugfixes

select MIDI device, if there are multiple
disable the WLAN AP correctly, if not used
don't initialize the OLA-universe (and throw an error), if it's not available

moritzvieli, 2024-07-07:
@DavidOpgh Thanks for the hint! Yes, the update was not fully rolled out for old versions. Can you please check again, whether you see the new version now?

DavidOpgh, 2024-07-09:
@moritzvieli - Version 2.3.6 update was seen by the PI 4 now. Thanks!