Rocket-Show/rocketshow #109: Extension to 16 Audio Channels

EkimRemarc, 2024-02-12:
RocktShow seems to have the right DNA compared to other multitrackers I have been used since years. 

For Me RocketShow needs two extensions to get more value:

First: I would like to use 16 Audio Channels which enables to have backings for each musician + orchestra in stereo plus a master click & band click. (This is not working in 2.3.5 tested with Behringer X32)

Second: I would suggest to implement a softswitch which globaly enables / disables audio busses. Reason is when a real musician is missing I must switch on his tracks for all compositions for a complete GIG just one time instead of modifying all compositions.

DavidOpgh, 2024-04-05:
Hello! RS does have a lot of nice features compared to other multirackers I've tried.

1) I've only used a stereo channel audio with RS. What happens when you try to connect it to the X32?

2) I would think this would easier to implement directly from from X32 than from RS.

moritzvieli, 2024-04-16:
@EkimRemarc Regarding Behringer X32: Same question from my side. We also use a Behringer X32 with 3-4 channels and this works. Does it only work until a specific number of channels? Did you check the 8-8 routing table?

Regarding the 2nd requirement: We're already discussing similar features here: https://github.com/moritzvieli/rocketshow/issues/96

As @DavidOpgh already mentioned: I'd also suggest to map all channels and just mute them in the X32 as required. This is also, what we do live: For 1-2 special songs with guest musicians, we just added separate backing tracks in a channel and mute them, when they're live with us on stage.

EkimRemarc, 2024-04-20:
It is working up to a certain amount of channels (8). You can configure more but they will not routed to the Behringer and RocketShow shows up a failure message starting while configuring the 9th channel. Up to 8 channels it works fine.

moritzvieli, 2024-06-03:
@EkimRemarc Can you show your routing configuration in the Behringer X32? Can you post the errormessage here (screenshot or download the logs from the Rocket Show settings)?

DavidOpgh, 2024-08-08:
@EkimRemarc did you get this issue resolved? Behringer had a sale and I bought an XR18 to test Rocket Show with more than 8 channels.

DavidOpgh, 2024-08-30:
I've started looking at this issue and I cam confirm RS has an issue with trying to assign more than 8 channels from my xr18

[image]

[image]

[image]

[image]

moritzvieli, 2024-09-02:
@DavidOpgh Just to be sure: I know, you route audio from the interface in the X32 in pairs of 8 channels. Did you set up routing in the X32 to more than 8 channels?

E.g. something like this?
[image]

DavidOpgh, 2024-09-02:
[image]

[image]

[image]

[image]

[image]

[image]

DavidOpgh, 2024-09-02:
[image]

[image]

[image]

DavidOpgh, 2024-11-11:
@moritzvieli have you been able to reproduce this error on your end?

moritzvieli, 2024-12-16:
I got my hands recently on an X32 rack version and could reproduce the issue. I have not found a solution yet. Not sure, what's causing the problems. I need to check, whether Gstreamer (Rocket Show's playback backend) is an issue or the X32 on Linux maybe.

DavidOpgh, 2024-12-16:
Thanks for the update.

DavidOpgh, 2024-12-16:
@moritzvieli 
I did a quick Google search. Did you see this? It seems to be related if not exactly the same. 
https://stackoverflow.com/questions/61728296/gstreamer-multichannel-greater-than-8-channels

DavidOpgh, 2024-12-16:
@moritzvieli 
This seems to indicate that there was a hard limit of 8 but was increased to 64 if using a newer version of gstreamer. 
https://stackoverflow.com/questions/68804159/gstreamer-input-from-osxaudiosrc-with-more-than-8-channels

moritzvieli, 2024-12-21:
> @moritzvieli This seems to indicate that there was a hard limit of 8 but was increased to 64 if using a newer version of gstreamer. https://stackoverflow.com/questions/68804159/gstreamer-input-from-osxaudiosrc-with-more-than-8-channels

This seems to be only related to OSX. I don't think, this fix should change something on the Raspberry Pi. Also, it has been released in FEB 2022. It should have been included on my test raspi V4, which has an image built in JUN 2023.

moritzvieli, 2024-12-21:
> @moritzvieli This seems to indicate that there was a hard limit of 8 but was increased to 64 if using a newer version of gstreamer. https://stackoverflow.com/questions/68804159/gstreamer-input-from-osxaudiosrc-with-more-than-8-channels

Yeah, also seen this one. Maybe I need to fiddle around with those caps, although it should not be necessary.

I'll try with a Raspberry Pi 5 or an updated version of Gstreamer. Maybe it's already fixed.

DavidOpgh, 2024-12-21:
> Yeah, also seen this one. Maybe I need to fiddle around with those caps, although it should not be necessary.
>
> I'll try with a Raspberry Pi 5 or an updated version of Gstreamer. Maybe it's already [...]

Thanks for the update. 64 might be excessive although I can see using more than 8 tracks of audio

DavidOpgh, 2025-01-20:
> Yeah, also seen this one. Maybe I need to fiddle around with those caps, although it should not be necessary.
>
> I'll try with a Raspberry Pi 5 or an updated version of Gstreamer. Maybe it's already [...]

@moritzvieli 
Were you able to test this?

moritzvieli, 2025-03-05:
@DavidOpgh I was able to test it yesterday evening but unfortunately, I got the same error with the new version. Need to dig further.

DavidOpgh, 2025-03-05:
> [@DavidOpgh](https://github.com/DavidOpgh) I was able to test it yesterday evening but unfortunately, I got the same error with the new version. Need to dig further.

OK - thanks for the update.