Rocket-Show/rocketshow #111: Rocket Show Startup Guide 2.0

DavidOpgh, 2024-07-09:
I created a new version of my Rocket Show Startup Guide.

Comments and corrections would be welcomed. 

https://docs.google.com/document/d/1qc-VDSoUHncOlFBArfpUA7hCuyIxQjfIOHbg5ceMGdw/edit?usp=sharing

moritzvieli, 2024-07-13:
@DavidOpgh Thanks a lot! This is great! We need to find a way to publish it properly. I'd still prefer it somewhere here on github. I already prepared https://github.com/moritzvieli/rocketshow/tree/master/docs for https://github.com/moritzvieli/rocketshow/issues/77. The advantage would be that everyone could access and edit it. If there's an easy way to convert Google Docs to Markdown, I think that's it.

Do you agree?

DavidOpgh, 2024-07-14:
I agree.  It should be on Github. 
Anyone should feel free to take it and convert it to Markdown. 
A Google search shows a couple different apps to do the conversion to Markdown.

moritzvieli, 2024-07-15:
@DavidOpgh I see, your guide is a little specific to your gear and contains some "i like", etc. The question is, do you want to release it e.g. as a PDF or something and I could feature it (and credit you of course) on the website directly? Somewhere in the getting started section? Something like "DavidOpgh wrote this excellent startup guide for beginners"?

I would still prefer it on GitHub as Markdown, because the chances of it being outdated would decrease and other people could easily collaborate (including myself). With a PDF version, you would need to release it regularly or I would have to drop it maybe somewhen in the future. But in order to add it to GitHub, we would need to rewrite some paragraphs.

Let me know, what you prefer.

Also, I think with your new startup guide, we can close https://github.com/moritzvieli/rocketshow/issues/77. Do you agree or is there something missing, you need to take over?

Thanks a lot for your feedback!

DavidOpgh, 2024-07-15:
The guide is somewhat specific to my gear but I tried to make it generic
enough for most people.

I'm OK with whatever format you want to use to publish it. I agree making
it a PDF doesn't allow it to be updated.
I'm not familiar with using Markdown. I played around with a Google Doc
to Markdown conversion extension but I didn't have a way to view the
result.

#77 can be closed.

DavidOpgh, 2024-07-16:
I used a Google Doc to Markdown conversion extension. 
The output is here. 
https://drive.google.com/file/d/1d-HxMY5CzauSvEWmOxChWAm2aTXoZW2I/view?usp=sharing

moritzvieli, 2024-07-16:
@DavidOpgh I just tried the same extension but the output was not that good and all images out of place. After downloading it as website and converting the HTML to Markdown with ChatGPT, it does not look that bad on the first glance, from my point of view:

https://github.com/moritzvieli/rocketshow/blob/master/docs/getting-started/getting-started.md

Can you also check?

I also made ChatGPT reword personal statements and restructure the markdown but large parts were just left out and i could not prompt it to include all parts. So this is what I would go with, for now, if fine for you. (y)

DavidOpgh, 2024-07-16:
I did a quick review.

One comment about the conversion.
I used an example in the OLA section with a simple DMX light but only 1/2 of it is included.  

This portion including the image wasn't transferred.
In this example I’m sending DMX information to the first 4 channels of this simple DMX light. The master dimmer (ch1=84), red (ch2=184), green (ch 3=181) and blue (ch4=169)

since I reference this example later I think this portion needs to be restored.

moritzvieli, 2024-07-22:
@DavidOpgh Thanks! I just added this missing part manually. Do you see any other issues? I linked the guide on the bottom of this page: https://rocketshow.net/get-started/

DavidOpgh, 2024-07-22:
I'll give the guide a detailed review to see if there are any other issues. What would be the process to make changes to the startup guide?

moritzvieli, 2024-07-22:
Thanks! Everyone, especially you, is welcome to make the changes to the documentation (or any other file) with a so-called fork and a pull-request. This is a standard-process in GitHub and can be used for most repos. This video gives a good explanation over how it works: https://youtu.be/nT8KGYVurIU?si=MxxDiDDgI0xfW7yP

The files for your startup-guide are all located here: https://github.com/moritzvieli/rocketshow/tree/master/docs/getting-started

Please let me know, if you have any issues. Otherwise I'd be happy to see Pull Requests from you. :) You'd then also see yourself as an official contributor to the Rocket Show repository finally (which is really about time, after all the good ideas, inputs and tests you already contributed over different channels ;)).

closed 2024-07-22 by moritzvieli