Rocket-Show/rocketshow #115: 2.3.8 New Backup/Restore function does not included Designer Project files, custom fixture profiles.

DavidOpgh, 2024-07-31:
@moritzvieli 
I tried using the backup/restore function in 2.3.8 to move my files from the Pi4 to the Pi5. 

Although the process looked like it was successful I'm not seeing the Designer Project files or my custom fixture profiles in either the backup file or the Pi5.

Is this an oversight or is there some other process to backup/restore these files?

moritzvieli, 2024-07-31:
@DavidOpgh Indeed. This was an oversight. Just released 2.3.9. Can you please try again? You first need to update both devices and create the backup again. Sorry for the inconvenience. Please let me know, whether it works now. Thanks a lot!

DavidOpgh, 2024-08-01:
@moritzvieli 
2.3.9 backup/restore function now includes the Designer Project files and and custom fixture profiles. I will close this issue.

I will note: 
on the Pi4 version the text labels for this function don't appear to be reference correctly?
[image]
[image]

One related issue is the Update window never closes after an update and reboot, you must manually refresh the browser, unlike other similar windows. 
[image]

closed 2024-08-01 by DavidOpgh