Rocket-Show/rocketshow #119: DMXking UltraDMX Max USB interface issue with OLA

DavidOpgh, 2024-08-22:
I thought I'd try upgrading my USB DMX interface so I bought the UltraDMX Max. 

OLA recognizes it, creates a universe, uses the "UART native DMX" plugin, but I can't get any it to send any DMX signals to a fixture. I've tried a combination of enable/disable plugins in but had no effect. Using the console page OLA "thinks" it's sending DMX signals but the fixture doesn't respond. 

I know the Max and the fixture are configured correctly by using the configuration utility software on my laptop. Using it's diagnostic page I was successful to send DMX signals on the expected channels. 

It's seems odd to use the UART name DMX plugin but I can't find a way to force OLA to use a different plugin.

I've logged an issue with DMXking support but haven't gotten a response yet.

DavidOpgh, 2024-08-23:
@moritzvieli 
I got this reply from OLA about my issue.
Where were is the  ola-usbserial.conf located?

Peter Newman <notifications@github.com>
10:37 AM (1 hour ago)
to OpenLightingProject/ola, DavidOpgh, Mention

Have a look at our FAQ @DavidOpgh :
https://www.openlighting.org/ola/get-help/ola-faq/#What_is_the_difference_between_the_different_USB_plugins

I suspect you've got the FTDI USB DMX plugin enabled rather than the Serial USB/USB Pro.

I assume this device is supposed to use the Enttec USB Pro protocols, I found it rather hard to tell from their specs.

Actually the device path/chip used will have probably confused things, as mentioned here:
https://docs.openlighting.org/ola/conf/ola-usbserial.conf.html

Try adding a line like this to your ola-usbserial.conf:
device_prefix = ttyACM

If that doesn't work please can you share your olad -l 4 logs as explained here:
https://www.openlighting.org/ola/get-help/ola-faq/#How_do_I_get_olad_-l_4_logs

DavidOpgh, 2024-08-24:
@moritzvieli 
Using WinSCP on my laptop I found  ola-usbserial.conf in etc/ola but I'm unsure of the process to update the file.  After editing ola-usbserial.conf I am unable to save the changes due to denied permissions even after shutting down OLA. I'm assuming I also need to shutdown RocketShow?

moritzvieli, 2024-08-24:
Try this command first:
sudo su - root

Does this help?

DavidOpgh, 2024-08-24:
Where do I type that command?

It didn't work when I opened a terminal window in WinSCP 
nor did it work when I typed it locally at the terminal prompt on the Pi5

DavidOpgh, 2024-08-24:
@moritzvieli - Never mind. I figured it out using the nano editor. 
Adding device_prefix = ttyACM to ola-usbserial.conf didn't fix the issue. 
OLA won't create any universe now.

moritzvieli, 2024-08-24:
Wait a second. The universe is created by rocket show. So the issue now is,
there‘s no universe at all? And you cant create it manually?

DavidOpgh, 2024-08-24:
That is correct. 

Once I added device_prefix = ttyACM to ola-usbserial.conf neither RocketShow  or I couldn't create a universe regardless of what plugins I enabled/disabled. 

I'm going to repeat the test because I had issues putting everything back to the original configuration. It seems the Max hung up when there was an active fixture connected to it. When I removed the active fixture the Max responded to changing the configuration back to the original setup. I was able to create the original "UART native DMX" universe when that plugin was enabled.

DavidOpgh, 2024-08-26:
This issue has been resolved. 
Adding device_prefix = ttyACM to ola-usbserial.conf fixed the issue with using DMXking UltraDMX Max USB interface with OLA.

What really compounded this issue was a new issue which took me a while to track down. I had a powered USB hub plugged into the Pi 5 but it's power supply got knocked loose. Even though the Max was not plugged into the hub it somehow caused the Max to hang up on a reboot. I would have to unplug/plug the Max to get it to respond after a reboot. Restoring the power to the hub or removing the hub entirely corrected this new issue.

DavidOpgh, 2024-08-26:
Peter Newman from the OpenLightingProject/ola github gave me the solution to this issue.

closed 2024-08-26 by DavidOpgh