Rocket-Show/rocketshow #125: Older IOS / Chrome or Safari unable to display RocketShow

rennist, 2024-10-31:
I have an older iPad I have been using for RocketShow.   It has maxed out at IOS 10.3.3  and because of that the highest Chrome version I can install is 71.0.3578.89.
It connected and displayed fine in version 2.3.2 but once I upgraded to 2.4.1 I can still connect to the RocketShow wifi but the browsers only show a blank page.  It is connecting to  rocketshow.local without error but just not displaying the pages.

Any idea why it stopped working with the 2.4.x versions ?  

Thanks

moritzvieli, 2024-11-01:
@rennist The required JavaScript version for the frontend has been increased by some major versions with this update. I just tried increasing the compatibility for older browsers with the latest update on the test channel. Can you check, whether it works again?

But even if it works, somewhen in the future, we would need to drop support for such old browsers.

Thanks a lot for your feedback!

moritzvieli, 2024-11-01:
@rennist not sure whether the new test version works. could be, that there are still issues. i just checked in more detail and maybe chrome this old will still not be supported anymore. Let me know, how it looks. 👍

rennist, 2024-11-01:
Hi @moritzvieli ... I just tested it and no, it still doesn't work.   Thanks for trying though.

closed 2024-11-01 by rennist