Rocket-Show/rocketshow #126: Play a composition from the composition editor directly

moritzvieli, 2024-11-18:
Try out a composition, without the need to save it first and switch to the play-tab.

moritzvieli, 2024-12-03:
@DavidOpgh @rennist This is now available on the test channel. Please let me know, if you find something. Thanks a lot!

moritzvieli, 2024-12-08:
@DavidOpgh I changed the volume sliders to a logarithmic scale to display volumes from -60 to 0 dB and pushed it to the test channel. I think it's fine now, based on the article (I had to read a couple more times to understand fully ;)). What do you think?

DavidOpgh, 2024-12-09:
@moritzvieli - The changes look good now. 
Unless @rennist has anything to add I think this issue can be closed.

rennist, 2024-12-09:
this working great in my testing so far.

DavidOpgh, 2024-12-09:
The only other thing I could think of that would be nice to have (although not absolutely needed) would be the ability to able to change the volume levels during playback.

moritzvieli, 2024-12-16:
This is now released.

@DavidOpgh Rocket Show is mainly used to playback preconfigured compositions. A volume control to adjust volume during playback could lead to unintended changes, if not secured properly with a warning message or behind another click somehow. If really required (which should not be the case currently, as you mentioned), please open a new issue. Thanks a lot!

closed 2024-12-16 by moritzvieli

DavidOpgh, 2024-12-16:
Thanks for the update.