Rocket-Show/rocketshow #127: Show Designer curve effect phase is relative to the time in the composition, not the corresponding region

moritzvieli, 2024-11-18:
I got an issue by email from Lars. Thanks!

The issue is described as follows: The curve effect has an adjustable phase. This way, the starting point of the curve can be modified. If a composition is played with a timeline, the phase is calculated based on the playing time and not on the region, the effect is placed in.

Example: The curve effect's phase is set to make sure, a curve always starts at the maximum (e.g. dimmer at maximum brightness). Now, different regions are placed on the timeline.

Currently, if the region is placed at time 0, the phase works as expected and the curve starts at the maximum. If it is placed later however, it might start at a different point.

What Lars suggests is as follows: Calculate the phase based on the placed region. This means, everytime a curve effect starts playing, it starts at the same position.

What do you think, @DavidOpgh?

DavidOpgh, 2024-11-18:
I've never played around with the curve effect but I agree with Lars. If the curve effect has an adjustable starting phase, calculate the phase based on the placed region. The curves should be able to start at the same position, regardless where it's located on the timeline. 

I can't think of a scenario where you wouldn't want that.

moritzvieli, 2024-11-18:
Ok, thanks for the quick reply! I'll implement it like this!