Rocket-Show/rocketshow #129: New Rocket Show Hardware

DavidOpgh, 2024-11-24:
@moritzvieli I see the website is previewing the new RS Hardware.
It looks very impressive, although I would have preferred XLR balanced audio outputs instead of RCA.
Was there a particular reason why you went with RCA?
 
Care to share what kind of hardware is inside?

moritzvieli, 2024-11-24:
@DavidOpgh jeah, I prefer balanced XLR as well, but seems like such audio interfaces are quite rare (or expensive) and people in this segment are used to RCA (see the ESI Gigaport EX for example). Regarding the hardware:
- it's a Raspberry Pi 4b with 2 GB ram (maybe upgraded soon) and 128 GB SD card
- Sound is provided by an Amanero I2S Interface equivalent (two times)
- DMX is a generic FTDI chip (learned it from you in another issue :))
- MIDI is provided by a generic chip as well

Maybe I'll open the specs and CAD designs for the case as well for users to build it by themselves.

closed 2024-11-24 by moritzvieli

DavidOpgh, 2024-11-24:
I have this but haven't tried it out with Rocket Show yet. 
https://www.amazon.com/gp/product/B0CX5K386N

moritzvieli, 2024-12-02:
@DavidOpgh Thanks for the tipp! Unfortunately, this does not ship to me. I checked other platforms but have not found a similar cable with USB A, only USB C. What's the quality of this cable? Maybe I'll use it in a future version.

DavidOpgh, 2024-12-16:
@moritzvieli 
Is there a reason you're not going with a Pi 5? 
I would think the difference in cost between the Pi 4 and Pi 5 would be minimal.
And is there a power supply included?

moritzvieli, 2024-12-21:
@DavidOpgh already decided to switch to a Pi 5. as you mentioned, because it does not cost much more.

moritzvieli, 2024-12-21:
> I haven't tried it yet so I can comment on the quality.
> 
> This is also this https://www.amazon.com/Devinal-Output-Converter-Amplifier-Speaker/dp/B08YNJ9XVD

Unfortunately also not available in my country. Unfortunately, Amazon does not ship much items to us.

DavidOpgh, 2024-12-21:
> @DavidOpgh already decided to switch to a Pi 5. as you mentioned, because it does not cost much more.

Excellent. The Pi 5 is significant improvement over the Pi 4